makerplane/pyEfis #12: Enhancement request: Alternative attitude indicator modes

Maker42, 2018-10-11:
A typical CDI needle abstractly represents a ground course beneath the aircraft.

Unfortunately it's so abstract it takes the concentrated development of skill to effectively follow it. One of the main problems is that deviation sensitivity increases rapidly as distance from the NAVAID decreases, and vice versa.

Modern EFIS displays have the opportunity to solve that problem by projecting 3D reality onto a 2D display and give the pilot a picture that much more closely represents a VMC situation (you can see what's outside with clear skies).

One way to achieve this is to project a simple course line onto the bottom part of the artificial horizon in such a way that it would look approximately the same if someone spray painted the route on the ground. The concept could be extended to display runway projections, IFR fixes, and even entire charts.

The specifics of the enhancement are:
A. Reality projection on artificial horizon, with overlay options that can be independently turned on and off:

- Runways
- Routes (VFR routes, IFR routes, Jet routes)
- Fixes / NAVAIDS
- Charts (select your chart type)
- Hoops in the air to fly through, representing an IFR approach (like projecting routes with vertical guidance in addition)

B. Transparent Attitude projection on a chart, looking straight down. This is like a moving map display, but you can keep an eye on the airplane attitude (and perhaps traffic info?) while you're studying the chart so the pilot doesn't let the aircraft go wild while his eyes are crossing trying to read the morse code pattern of the VOR.

neil-d95, 2018-10-11:
So there is an FAA standard in regards to CDI deflection in the different modes using a GPS.

1. Enroute Mode is 30nm outside destination airport is 5nm from the desired course at full deflection.
2 Terminal Mode  is inside 30 nm of destination 1 nm from the desired course at full deflection
3. Approach Mode is within two nm of the final approach fix, CDI sensitivity scales to "APR" mode, in which a full-scale CDI deflection indicates at least 0.3 nm from the desired course.

I'll have to look up the FAR to get the exact wording.

Maker42, 2018-10-11:
Wow Neil, you are a walking encyclopedia of avionics FARs!
To be clear, this enhancement request does not propose to change any representation we make of a CDI, but rather offer a "virtual VFR" supplemental information view.

closed 2018-10-29 by Maker42