makerplane/pyEfis #13: Enhancement request: Show air data in numerical form

Maker42, 2018-10-12:
In addition to the airspeed, altitude and heading sliders or round gauges, display the actual number.

https://github.com/mcvsama/xefis has a good example of this.

birkelbach, 2018-10-13:
I think this has always been in the plans, we just haven't gotten to it yet.  I like the way most of them 'roll' the numbers like an old odometer.  It helps give a since of rate of change on the numbers as well.

neil-d95, 2018-10-13:
So I was just looking and noticed the IAS number and the Alt Settings are no longer working the IAS allowed you to bounce between IAS TAS and GS guess that is a fun bug to go fix. I understand about having the numbers next to the line. I thought I had a version that did something like that but it's been awhile.

birkelbach, 2018-10-14:
I'm not sure what happened to those little boxes.  I think the widgets are still there but maybe they got un-configured when Jean-Manuel did all that work for his Cub.  How about a little box that shows OAT, Density Altitude, TAS and GS.  I have a little box like that on my Dynon D10 and I like it.

The little IAS and Altitude boxes are a little tricky because of they way those widgets are drawn but I think  it's worth doing well.

neil-d95, 2018-10-14:
Yeah, That was the idea when I did the IAS and the concern I have is 
processor usage and memory size. I'll work on getting everything fix 
then try the boxes again I know I did something similar awhile ago.

On 10/13/18 10:01 PM, Phil Birkelbach wrote:
> m not sure what happened to those little boxes. I think the widgets
> are still there but maybe they got un-configured when Jean-Manuel did
> all that work for his Cub. How about a little box that [...]

closed 2018-10-29 by Maker42