makerplane/pyEfis #41: Disappearing Menus

birkelbach, 2019-01-28:
We had talked about making the menus only appear when a button was pressed, and then disappear after a certain timeout.  I think it'd like it to be even more sophisticated than that.  Say I already know what a button does without the menu so I press that button.  Let the thing go ahead and do it.  Say switching screens.  I'd probably have those memorized so instead of having to press a key to get it to show up and then press another to get it to do the thing that I want it can just go ahead and do it.  Maybe one of the buttons could be to 'bring up the menu' and the rest just do their thing.  This could be a configuration thing so that each user could customize it to their liking.  Just throwing out ideas but I do think the menu should only appear when needed.  At least be configurable to do so for the smaller screens.

makerplane-jnicol, 2019-01-28:
Configurable in the settings is good.  I tend to agree about knowing what the buttons do after a short time using it.  For deeper menu items though that aren't used a lot, this might be harder.  Maybe it could be a double press of the left rotary encoder, or maybe two buttons at once?

birkelbach, 2019-01-28:
It might even be as simple as having a button attribute that says "function while invisible = True" and maybe another one that says "menu visibility trigger = True"  If both are true then the button would work and bring up the rest of the menu.  If the first one is false then it would just bring up the menu and the second press would perform the function.  If the user wanted to have a single button do nothing but show the menu then it'd be simple enough to do.  For very simple systems the buttons might also be labeled externally (silk screen) so no menus would be needed.  But the button -> feature assignment would still need to be there.  I don't really think that externally labeled buttons is necessarily a good idea, I'm just thinking out loud.

birkelbach, 2019-01-30:
What I'm asking for is that the entire menu disappear (visibility = False) when not being used.  Screen real estate is valuable on the smaller displays.

makerplane-jnicol, 2019-01-31:
Good idea.

Links:
------
[1] https://github.com/makerplane/pyEfis/issues/41#issuecomment-459035597
[2]  
https://github.com/

Maker42, 2019-03-26:
Hi Phil,

Are you making progress on this, or would you like me to pick it up?

birkelbach, 2019-03-27:
I haven't worked on this.  I figured the menu was your baby.

closed 2019-05-01 by neil-d95