makerplane/pyEfis #42: Distribution

birkelbach, 2019-01-28:
At some point we need to refactor the directory structure and get this thing in some kind of shape so that it's easy for users to install.  Whether that be using PIP or something else would be up for discussion.  I did this for FIXGateway and it's not trivial, so it should probably wait until after OSH19.

MagnusBieneck, 2023-02-16:
Hi @birkelbach, I would like to give it a try, as my first contribution. Do you have anything else I should consider, other than that it should be released on PyPI?

birkelbach, 2023-02-16:
When I did this for FiXGateway it made life much easier to follow the guidance in the PIPs for structuring the project.  I was just learning Python when I started this project so it's a bit of a disaster.  Give it a go, welcome aboard!!

closed 2024-03-01 by e100