makerplane/pyEfis #49: Value Quality Display for Instruments

birkelbach, 2019-01-30:
We need to make sure we display bad quality data to the user in some way.  Either by flags or graying out the information etc.  There are three flags "Old", "Bad" and "Failed"  Old means we haven't gotten data within the proper timeout period if given.  Bad means the device sending the data has reason to believe that the data may not be as accurate as it should be, but that the data is still available.  One example might be a device that is still working but one of the components is operating outside of it's temperature range.  Failed means that we KNOW the data is wrong or missing.

Old and Bad data can still be displayed but grayed out or flagged in some way.  Failed data should not be indicated.

The gauges already do this but we need to add it to other critical instruments like the AI, heading, airspeed, altitude etc.

makerplane-jnicol, 2019-01-31:
Agree.  Are you looking at the standard red cross through the  
instrument to indicate failed?

Links:
------
[1] https://github.com/makerplane/pyEfis/issues/49
[2]  
https://github.com/

birkelbach, 2019-01-31:
> Agree.  Are you looking at the standard red cross through the instrument to indicate failed?

If it makes sense.  It probably depends on the instrument.  The gauges replace the value text display with Xs and the pointer disappears.  I'd think something similar for airspeed, altitude, heading etc..  For attitude, turn rate, slip skid and more graphical indications maybe go grey on bad data and turn the whole instrument black on failure.  Just my thoughts.  I do think it's important for the attitude indication to go away if we know it's bad data.  Many a pilot have died with a red flag hanging down on the corner of their artificial horizon, we can do better than that.

spaschke64, 2019-01-31:
I would urge you not to do what Aspen did and black out the whole screen if the pitot is plugged. There is still good data coming from gps, gyros and static lines. 

Stephen Paschke
720-236-8717

closed 2019-05-01 by neil-d95