makerplane/pyEfis #50: Mirrored Display

makerplane-jnicol, 2019-01-31:
It would be useful to enable mirroring of the display for HUD applications.  Future enhancement.  See the Stratux Hud project.  https://github.com/JohnMarzulli/StratuxHud

https://www.youtube.com/watch?v=Du7jkJTLTPo

makerplane-jnicol, 2019-01-31:
> That's pretty cool. We need to talk to those guys about getting it
> integrated. Mirroring the display is one thing but I'd think we
> might want a different screen displayed on the HUD. Perhaps the
> ability to show two screens on two different outputs at the same time.

Yep, I meant mirroring as a general UI configuration and allow folks  
to create their own HUD display.  Potentially re-using the displays  
already created by this project.  At a minimum, they could just put up  
the mirrored PFD.  I guess the screen may need to be flipped  
vertically as well as an option.

Links:
------
[1] https://github.com/makerplane/pyEfis/issues/50#issuecomment-459339771
[2]  
https://github.com/

birkelbach, 2019-01-31:
Roger.  We're on the same page.  I like it.  We need to talk to the Stratux guys at Airventure.

makerplane-jnicol, 2019-02-01:
For interest and technical information on HUDs here is an interesting web post by a former CTO of a HUD company.  https://www.kguttag.com/2018/01/29/ces-2018-part-2-aftermarket-heads-up-display-hud/

The Hudley is what was used in the StratuxHUD project.

spaschke64, 2019-02-01:
I purchased some lightly mirror tinted window film to experiment with.  I was thinking I would set up a car visor shaped clear piece of plastic with the film. Put it on 
a moveable arm. Try it out with my phone to see if it is bright enough. The small portable led projectors would be if the phone isn’t. 

Stephen Paschke
720-236-8717