makerplane/pyEfis #51: Event System

birkelbach, 2019-01-31:
Hey guys.  What do you think about this little feature...

[events](https://github.com/birkelbach/pyEfis/tree/events)

What it boils down to is a way to send global signals in the program with strings instead of having to program them into a module (i.e. hooks) or eval() code like the menu module does.  I needed a global event dispatching system for the EGT monitor for setting leaning modes and sending commands to normalize and reset, etc.  I worked up a few methods and this is where I am so far.  The nice thing about using the strings is now we can let the user setup key bindings with configuration like this...

[Keys]
A = Show Prev Screen
S = Show Next Screen
P = Show Screen, PFD
E = Show Screen, EMS
M = Set Airspeed Mode, TAS
...

It would also let us replace... 

"self.myparent.change_asd_mode_event (QEvent(QEvent.ActionAdded))"

... in the menu configuration with...

"Set Airspeed Mode, Next"

...or something similar.

What do y'all think?  It still needs some work on how it's imported and I still need to figure out how to handle the key bindings but this is the direction that I'm headed.  This pretty much replaces the "hooks" module and I'll probably recommend that we delete that unless you guys can think of a reason to keep it.

makerplane-jnicol, 2019-01-31:
I guess it would allow folks to use programmable usb keyboard interfaces like this one?:https://www.ultimarc.com/ipac1.htmlThis kind of thing has been used for flight simulators to bind switches and buttons to key presses.Not sure if this is where you were headed, or if it is something that you would use through pi GPIO?

birkelbach, 2019-01-31:
It could be either.  It could also be fired from external data.  The main thing about this event module is not so much about "why" to fire the events but "how."  All it really does at this point is convert strings to PyQt Signals.  This just seems easier from a configuration standpoint.

spaschke64, 2019-02-01:
That board looks like an easy way to hook up a rotary encoder

Stephen Paschke
720-236-8717

makerplane-jnicol, 2019-02-01:
Hey Steve, welcome.These boards are also used for mame cabinets as well as simulators, so you can hook up a lot of things.  If Phil implements key bindings then I guess it will open up a lot of different hardware solutions like this if they are not implemented directly in an SBC like a raspberry pi.John

birkelbach, 2019-02-04:
I did some more work on this.  I added keybindings and juggled the file structure around a little bit.  Instead of 'events' I called it 'hmi' for Human Machine Interface.  I moved Garrett's menu module into the hmi package, and I'm planning to add a feature where the actions can be triggered on database value changes as well.  Then external buttons can fire actions as well.  Current code is at... [hmi branch](https://github.com/birkelbach/pyEfis/tree/hmi).  Imagine being able to manipulate the menu of your EFIS with buttons on your stick grip.

makerplane-jnicol, 2019-02-06:
Hey Phil, 
One more thought is to add in the ability for a key to swap display screens from one pyEfis to another.  Should I put this in as a new Feature Request?

neil-d95, 2019-02-06:
?

makerplane-jnicol, 2019-02-06:
So if you had two physical displays set up in the cockpit.  One could be the PFD and the other the moving map.  Hit the 'swap display' button and it swaps the map to the pfd and vice versa.  It would probably mean setting up one as the primary display.

birkelbach, 2019-02-06:
Oh Okay. I don't think it needs to be two pyEfis's running just one that puts different screens on different displays.  Switching the two would be easy enough.  I think this is a feature that would come naturally when we get around to deciding how to support multiple displays.

birkelbach, 2019-02-21:
Fixed by #62

closed 2019-02-21 by birkelbach