makerplane/pyEfis #55: Historical temperature plot

Maker42, 2019-02-04:
We should be able to plot historical temperatures for CHT & EGT. Often temperature trends are more important than the absolute number. So the plot should also be able to highlight any place the absolute value of the first derivative exceeds a threshold.

Maker42, 2019-02-04:
With the ability to mute rapidly rising temperatures during runup and takeoff, or tolerate a higher threshold.

birkelbach, 2019-02-04:
I agree.  I was planning on adding this too but it's going to be hard to get it to fit on the current EMS screen, so I'll probably have another EMSTrend screen or some such thing.  It's pretty low on the priority list right now but it might get put in for Airventure.  I have an ems branch on my repo that is where I'm tracking my progress if you want to go take a look at the direction that I'm headed.  Right now all the work I'm doing is for the small screen to fit the stuff that Kyle is working on.

Maker42, 2019-02-04:
One way to do this is to have a strip of screen real estate at the bottom that can be used for various selected functions, while the top remains displaying the key information.

Maker42, 2019-02-04:
Hmmm... Yes, very limited. Maybe a bottom strip temporal popup functionality is more appropriate. So it temporarily and partially covers the bottom of the AI, but still leaves the horizon line untouched in straight and level flight.
If you're not straight and level, you probably won't want to be looking at CHT plots anyhow.

birkelbach, 2019-02-15:
Might consider this... https://github.com/pyqtgraph/pyqtgraph

Maker42, 2019-02-15:
That has potential. Of course what has to be weighed in the balance is its value add vs. the cost of yet another dependency. Unfortunately the documentation doesn't show any sample images, so it's hard to tell how much value it would add compared to Qt's drawLine. It also has a lot of bloat we don't need in the form of GUI prototyping and flow chart code.

birkelbach, 2019-02-15:
Mostly I put that there to remind me. I’m actually leaning toward our own solution. I’ve done these kinds of things before and it’s not terribly difficult.