makerplane/pyEfis #58: Bug: pyEfis crashes if FIX server is not running

Maker42, 2019-02-14:
It should either exit gracefully with a clear error message, or wait for the server to come alive, with a clear message as to why it's waiting.

Maker42, 2019-02-14:
In addition, it would be nice if the fix module provided a function to block until the entire database has been populated.

birkelbach, 2019-02-14:
pyEfis does not crash if the FIX-Gateway server is not running.  The issue is the initialization of the internal FIX database within the EFIS.  The database is initialized by calls to the get_item() functions in the individual instruments or systems that need the points.  Once the server comes alive, pyEfis starts getting data from the server and off we go.

There are really only two alternatives that I can think of.  We can predefine all of the database points and populate the database before anything else runs.  I see this as a maintenance issue.  We'll have to make sure that our pre-definitions match the FGW server and the instruments that we include in our configuration.  Or we can decide that the server must be up and running before the EFIS starts, then initialize the database from what we find in the server, and crash or wait if the server isn't running.  The problem with this is what do we do if the server crashes after we are up and running.  Now, if the server crashes all of the database points get marked as 'failed' and when the server comes back, they recover if the points are there.  If we have a requirement that the server be running then we'll have to crash the EFIS when the server crashes or figure something else out.  Another problem with initializing from the server is that we may create hundreds of data points that we don't care about.

I really don't see what the problem is letting the instruments and systems initialize the database while they are being initialized.  It seems to work pretty well right now as long as the initialization routines call get_item() with the create=True flag set if the function is being called in the initialization routine.  If it's not then leave the flag off and catch the KeyError exception and handle appropriately.

Maker42, 2019-02-14:
create=True works OK I guess. It just appeared to me a little off to say if data is not available, make something up. This fails hard when the expected data type or valid ranges are in conflict with the default, which is a real drawback.

It should be real easy to allow pyEfis to wait until the database is populated through the FIX server connection.

birkelbach, 2019-02-14:
When the pyEfis connection to the server is made it attempts to read all of the information in it's database.  It then sets all of the values, ranges etc.  The efis database will always match the gateway server database.  It will even change if the information in the server changes.  It's up to the individual consumers of the information to deal with these changes.  

It's done this way because I found that it is NOT real easy to wait for the server to start if we want the efis to survive the server failing.  Otherwise how do we go back and reconfigure the information if it changes.  Maybe this is a non issue but there is a lot of code that will have to change.  You also have to remember that a lot of this information comes from outside the system on the CAN bus.  Now we would have to have the server wait for all of the nodes to start, and it doesn't even know how many nodes to wait for.

Maker42, 2019-02-14:
What I'm talking about is upon initialization in pyEfis, there is a wait built in for all of the fields pre-configured in the FIX gateway to be sent to the fix modules database (config/default.db). This is a known finite list of items. For this pre-configured list, there is no dependency on outside nodes, so I don't see what the problem is here.

This does not effect any ongoing operation of the pyEfis, so the gateway crashing after the EFIS is up and running would not change anything. pyEfis would continue to run as it does now, so I fail to see the problem you're trying to point out. Can you make it clearer?

birkelbach, 2019-02-15:
I guess I don't really have a problem with a wait but it doesn't change the fact that consumers within the EFIS need to be very careful to watch for changes in the data.  This would make some of the problems that I'm having with the data bindings in the HMI work better.  We have to know that if the gateway server is shut down and restarted with a different database then what we are talking about will break.  I guess that is not likely and probably not worth worrying about.  I have some ideas.  Let me play with it and see what I can come up with.  The more I think about it the more I'm realizing that it would help with integrity checks of the data that we expect on startup.

birkelbach, 2019-02-16:
I just submitted a PR for this feature.  It was actually pretty easy.  The db.get_item() function will now wait until all of the database has been read from the FIX Server and the pyEfis database has been initialized with this data.  The get_item() function will now take the argument 'wait = False' if you want the old behavior.  It might make sense to have some internal database items that are not in the gateway.  Other than that, it acts the same.  If create=True and wait=False then it'll create the point and move on.  If create=False it'll raise a KeyError if not found regardless of the wait flag.  It still seems to handle server restarts okay but I don't know what it'll do if the server restarts with a different database.  Probably the same thing it used to and I can't imagine a scenario that would require the database to change in flight.  Mostly it should handle restarts of the server in case we restart it due to failure.

PR #60 (2019-02-20): Add a wait state for database initialization
Calls to db.get_item() will now wait until all of the database items
have been created based on the list from the FIX server.

Fixes #58
closed 2019-02-20 by neil-d95