makerplane/pyEfis #68: Optimize engine gauges

Maker42, 2019-02-26:
The currently checked in engine gauges repaint the entire gauge for every update/paint event in Python. This method has been shown to be very expensive based on profiler data.

If the more static parts of the gauge were in a graphicsview, and only the dynamic parts were repainted, there would likely be a large performance boost.

birkelbach, 2019-02-26:
This is on the short list.

birkelbach, 2019-08-16:
I've done some experiments with this and I have not been able to get much of a boost.  The issue is that the text is expensive to draw in the QGraphicsScene.  I got about a 15-20% boost without aligning the text.  When I right aligned the text it was actually slower.  This was because the position has to be calculated on each update.  Without the alignment it wasn't too bad but I couldn't figure out how to right align text without calculating a new position each time the text changes.  There might be a way.

There are probably some other optimizations we can do with the QWidget Gauges like making the static things a QPixMap or something instead of drawing them.  I'm not sure, but I'll play with it some more.