makerplane/pyEfis #103: Add Brightness Slider or buttons

makerplane-jnicol, 2021-02-07:
We will need to add in a brightness control for specific control of the GPIO PWM for the Pi.  I know we have also discussed a global brightness control that will dim or brighten the whole panel.   Right now we cannot control the brightness of the display from within pyEfis.

Code supplied by Steve to control this follows:

```
import RPi.GPIO as GPIO
import time

GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.OUT)
pwm=GPIO.PWM(18, 100)
pwm.start(0)

for dc in range(0,110,10):
        print dc
        pwm.ChangeDutyCycle(100-dc)
        time.sleep(1)

pwm.stop()
GPIO.cleanup()
```

e100, 2024-05-08:
The UI portion of this would be easy to setup with just configuration by using a vertical or horizontal bar gauge with an 'up' and 'down' transparent button on top of the gauge.

We would need a plugin in fix gateway that would set the screen brightness based on the value changed by the buttons.

birkelbach, 2024-05-09:
There is a "dimmer" plugin for FiXGateway.  It writes to a Linux device file that sets the screen brightness.  The MakerPlane 5 in display may not play nice with the Linux driver and therefore needs the above code.  I'm not sure.  I'll try to find some time to play with it.

e100, 2024-05-09:
I created an example dimmer control and submitted a pull request #189 
To add this to a screen one would just need to include this control on their screen definition, this can be done like:
```
  - type: include,includes/vertical_dimmer_control.yaml
    row: 90
    column: 16
    span:
      rows: 20
      columns: 6
```

I did not merge the pull request because this also requires adding some new fixids to the fixgateway and without some mechanism to auto-update default configuration files this could be a breaking change for some users.
I created #190 to discuss how to deal with the config files