makerplane/pyEfis #252: Installation of pyEfis on Windows

aiesha-khawaja, 2025-01-28:
I came across the PyEFIS library and would like to know if it supports installation and operation on a Windows environment.

Specifically:

Are there any known limitations or dependencies that might prevent installation on Windows?
If installation is supported, are there specific instructions or steps for setting it up on Windows?
I would appreciate any guidance or documentation related to this.

Thank you!

JanneMantyharju, 2025-01-28:
I haven't tried, but I don't think there are any major obstacles. Some file paths may be an issue. On MacOs Pyefis works fine. The question is why would you want to do this. WSL (Windows subsystem for Linux) works really well if you want to develop on Windows environment. Running the PyEfis on a plane on Windows is a bit fruitless since you likely need another computer for data acquisition anyway.

aiesha-khawaja, 2025-01-28:
Thanks a lot for your quick reply, I will try installing it using WSL.

aiesha-khawaja, 2025-01-28:
Another thing I want to ask is about "Update the config file [Screen.PFD] section dbpath and indexpath with the path names of the FAACIFP18 and index.bin files respectively."
If I have downloaded the FAA CIFP file and created the index.bin file, where is the [Screen.PFD] section, actually I want to make sure instead of the default data it only picks the data in "pyEfis/CIFP directory" I downloaded

e100, 2025-01-28:
@aiesha-khawaja 
This is definied in the config file config/screens/virtualvfr_db.yaml
By default it will look in:
dbpath: ~/makerplane/pyefis/CIFP/FAACIFP18
indexpath: ~/makerplane/pyefis/CIFP/index.bin

My suggestion is to put the files where it expects to find them.

It could run on windows I suppose but I am quite sure it will not mostly because of file path differences between windows and linux.

closed 2025-01-28 by e100