makerplane/pyEfis #255: Default snaps for fixgateway are failing importing Numpy

EdB987, 2025-02-22:
I've been cleaning up my install and wanted to try CarPiHat hat for a more reliable power on/power off solution because the X729 + RPi5 simply gets itself into a confused state if power cycled too many times. It is impossible to resolve in flight because the pi is behind my seat. In any case, I want the unit to power on and off **reliably** with a switch.

That being said, I have opted for a completely clean install. It seems that the current production fixgateway snap is failing when loading the mgl plugin and mgl wants to import Numpy. There is a dependency on libblas.so.3, which is not present in the build.

**Steps to reproduce:**

Enable mgl (not mgl_serial) and run fixgateway manually with --debug set:

fixgateway.server --config-file=$HOME/makerplane/fixgw/config/default.yaml --debug

**output is:**

CRITICAL:20250222-12:09:06:root - Unable to load module - fixgw.plugins.mgl: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.
Traceback (most recent call last):
  File "/snap/fixgateway/223/usr/lib/python3/dist-packages/numpy/core/__init__.py", line 24, in <module>
    from . import multiarray
  File "/snap/fixgateway/223/usr/lib/python3/dist-packages/numpy/core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/snap/fixgateway/223/usr/lib/python3/dist-packages/numpy/core/overrides.py", line 8, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libblas.so.3: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/snap/fixgateway/223/usr/lib/python3/dist-packages/numpy/__init__.py", line 130, in <module>
    from numpy.__config__ import show as show_config
  File "/snap/fixgateway/223/usr/lib/python3/dist-packages/numpy/__config__.py", line 4, in <module>
    from numpy.core._multiarray_umath import (
  File "/snap/fixgateway/223/usr/lib/python3/dist-packages/numpy/core/__init__.py", line 50, in <module>
    raise ImportError(msg)

**Further down:**
Original error was: libblas.so.3: cannot open shared object file: No such file or directory

e100, 2025-02-23:
If you run `sudo snap refresh` it should update fixgateway and hopefully work OK now.
I updated the LD_LIBRARY_PATH env var so it can find libblas.so.3

To tell if you are running the fixed version run `snap list`, the Rev column should report 226 (amd64) or 227 (arm64) for fixgateway, or possibly a higher number if other changes are pushed after I post this message.

EdB987, 2025-02-24:
Works perfectly. Thank you.

closed 2025-02-24 by e100